paulscherrerinstitute/StreamDevice #9: Packaging as a standalone EPICS module

darcato, 2018-05-14:
Is there a specific reason why this must be put inside the application top folder? All the other epics modules are distributed as standalone support library to be linked by the application?
Can this be done here? I know it can easily be done by the user, but this change would help to standardize the way a user installs an epics module.

keenanlang, 2018-06-06:
>Can this be done here?

Late response, but the APS does have a packaging of streamDevice as a standard epics module.

https://github.com/epics-modules/stream

darcato, 2018-06-07:
Thanks, I completely missed it. 
I will use that repository!

dirk-zimoch, 2018-06-07:
I'm working on it. Version 2.8 will come as a standard epics module. At 
the moment I am still working so some other improvements...

Dirk

dirk-zimoch, 2018-08-14:
Done in the current master branch.

closed 2018-08-14 by dirk-zimoch