paulscherrerinstitute/StreamDevice #11: Dynamic compiling on Windows gives unresolved symbols

keenanlang, 2018-06-07:
Dynamic build of StreamDevice on Windows (win32-x86, Visual Studio 2015) results in the following unresolved external symbols:

devaoStream.obj has unresolved symbol interruptAccept
devwaveformStream.obj has unresolved symbol pamapdbfType
StreamEpics.obj has unresolved symbol pdbbase

This is not encountered when I built against commit 76115ae

dirk-zimoch, 2018-06-08:
Hmm... These are symbols from base. Strange. I never use Windows and don't understand how the dll linking rules work. I need to set up a Windows machine with EPICS to reproduce. May take a while.

dirk-zimoch, 2018-06-20:
Should work now.

dirk-zimoch, 2018-06-20:
Note that I am working on the 2-8 branch at the moment. This is probably the last commit on the current master branch, which will be renamed 2-7 soon and 2-8 will become the new master.

closed 2018-06-20 by dirk-zimoch