paulscherrerinstitute/StreamDevice #12: No error message, if regular expressions are used, but PCRE is not configured

bl0x, 2018-07-19:
StreamDevice should fail when reading (or at least, when using) protocols that make use of regular expressions, even though PCRE is not configured and not compiled in. At the moment it seems to silently do nothing instead.

dirk-zimoch, 2018-07-19:
You don't get the message: "No converter registered for format '%/'" ? I will check that. Need to rebuild without PCRE first...

"At the moment" means which version/commit?

bl0x, 2018-07-27:
I was running with version 2_7_11. Sorry for the delay.

dirk-zimoch, 2018-07-27:
I had switched off error messages by default in 2_7_8 because too many people complained about "message flooding" when their device is offline. But that was not a good idea, because now you could not see the problem easily (e.g. protocol syntax errors). In a later version (2_7_12) I have re-enabled error messages during initialization. 
With your version, you may want to set 'var streamError 1' in your startup script. When done before iocInit, you should see messages about the unknown format. At run time you should see "Record not initialised correctly". 
In any case the SEVR/STAT fields of the record show the error. In this case INVALID/UDF. Any graphical user interface that shows the alarm status (which I always recommend) would show the INVALID alarm.

At the moment I am working on version 2_8_0 and I am trying to find a better balance between message flooding and problem hiding.

dirk-zimoch, 2018-08-14:
Fixed. Error messages are now always on during initialization.

closed 2018-08-14 by dirk-zimoch