paulscherrerinstitute/StreamDevice #18: configure/RELEASE SUPPORT definition

keenanlang, 2018-09-21:
synApps generally doesn't use full path names for referencing each module, instead defining module paths in reference to the macro SUPPORT. The use of the command 'make release' with StreamDevice ends up writing ASYN et al. as $(SUPPORT)/<module>-<tag>, but since configure/RELEASE doesn't have the SUPPORT macro, when trying to build, the modules aren't able to be found.

Even just adding "SUPPORT=" alongside the EPICS_BASE definition would be enough to get things to work with make release, even if you don't define it as anything or use said macro.

dirk-zimoch, 2018-09-25:
I have now added SUPPORT= to configure/RELEASE

closed 2018-09-25 by dirk-zimoch

keenanlang, 2019-05-28:
The line you added to the release file is "RELEASE=" not "SUPPORT="

dirk-zimoch, 2019-05-29:
Oops. Fixed.