paulscherrerinstitute/StreamDevice #21: echo in Makefile breaks build on Windows

kmpeters, 2018-11-14:
This echo statement breaks builds on Windows:

https://github.com/paulscherrerinstitute/StreamDevice/blob/493dc19d8bb7880abdcae30339950512144a402e/src/Makefile#L91

The resulting stream.dbd file includes the quotes and an extra space:
```
variable(streamDebug, int)
variable(streamError, int)
registrar(streamRegistrar)
driver(stream)
device(ao,INST_IO,devaoStream,"stream")
device(ai,INST_IO,devaiStream,"stream")
device(bo,INST_IO,devboStream,"stream")
device(bi,INST_IO,devbiStream,"stream")
device(mbbo,INST_IO,devmbboStream,"stream")
device(mbbi,INST_IO,devmbbiStream,"stream")
device(mbboDirect,INST_IO,devmbboDirectStream,"stream")
device(mbbiDirect,INST_IO,devmbbiDirectStream,"stream")
device(longout,INST_IO,devlongoutStream,"stream")
device(longin,INST_IO,devlonginStream,"stream")
device(stringout,INST_IO,devstringoutStream,"stream")
device(stringin,INST_IO,devstringinStream,"stream")
device(waveform,INST_IO,devwaveformStream,"stream")
device(aai,INST_IO,devaaiStream,"stream")
device(aao,INST_IO,devaaoStream,"stream")
device(calcout,INST_IO,devcalcoutStream,"stream")
device(lsi,INST_IO,devlsiStream,"stream")
device(lso,INST_IO,devlsoStream,"stream")
device(scalcout,INST_IO,devscalcoutStream,"stream")
"registrar(AsynDriverInterfaceRegistrar)"
```
And the build fails with the following error while creating streamApp.dbd:
```
dbdExpand.pl: Syntax error in '"registrar(AsynDriverInterfaceRegistrar)" '
Context: file '../../../dbd/stream.dbd'
  while reading 'stream.dbd' to create 'streamApp.dbd'
dbdExpand.pl: Exiting due to errors
```

kmpeters, 2018-11-14:
I should add that the build is being done in a cmd.exe window, which is why it uses the problematic echo.exe.

dirk-zimoch, 2018-11-15:
I will have a look. Maybe I used cygwin.

dirk-zimoch, 2018-11-15:
bash and powershell both need the quotes because of the parentheses. Only echo.exe requires no quotes. I don't know how to solve this dilemma at the moment.

dirk-zimoch, 2018-11-15:
I have re-built the project on Windows but don't have the quote problem. However \cygwin\bin is the first thing in my PATH. Also I use GNU make version 4.2.1 from cygwin if that matters. That does not mean that I build for cygwin though. I use the Microsoft cl compiler but everything else like make, perl, and as it seems echo as well, from cygwin.

kmpeters, 2018-11-15:
I use Windows reluctantly and I'm definitely not a Makefile expert, however, my perception is that putting ``registrar(AsynDriverInterfaceRegistrar)`` in its own dbd file and then optionally including that dbd file if ASYN is defined is significantly easier than trying to make the echo statement work for bash & cmd.exe.  This would require ``makedbd.pl`` to output an intermediate dbd file rather than the final one.

dirk-zimoch, 2018-11-16:
Fixed in commit 8f34dd2.
I have moved generation of the problematic line into the makedbd.pl script.

closed 2018-11-16 by dirk-zimoch