paulscherrerinstitute/StreamDevice #25: Comma separated strings not separating correctly

daykin, 2018-12-13:
Hi,

We are trying to accept a comma-separated response into a waveform of FTVL type STRING, but the strings are not separating correctly. The first 40 characters of the raw response are still being placed in element 0 of our record. If we modify our test server to use spaces, the waveform behaves correctly. 
I have a test case for this [here](https://github.com/daykin/StreamDevice/commit/531d058484ce20f83c93ad3afd28be3488d0090d).

Below is our use case -- maybe we're doing something wrong?

**Not-working protocol:**
Terminator = LF;
ReplyTimeout = 20000;
Separator = ",";
getVersions {
 out "GET VALUE VERSION AMP \$1";
 in "VALUE VERION AMP \$1 %s";
}

Server:
received "GET VALUE VERSION AMP 1"
sending response "VALUE VERSION AMP 1 1000,1.2.3,1010,4.5.6,1011,7.8.9,1012,10.11.12,1013,13.14.15,1020,2.4.6,1021,4.6.8,1022,6.8.10,1023,10.12.14\\n"
 

IOC:
epics>dbgf _FS1_CH01:RFA_D2137:FVSN_RD_ALL
DBR_STRING: "1000,1.2.3,1010,4.5.6,1011,7.8.9,1012,1"

**Working protocol**:
Terminator = LF;
ReplyTimeout = 20000;
Separator = "\_";
getVersions {
 out "GET VALUE VERSION AMP \$1";
 in "VALUE VERION AMP \$1 %s";
}
 

Server:
received "GET VALUE VERSION AMP 1"
sending response "VALUE VERSION AMP 1 1000 1.2.3 1010 4.5.6 1011 7.8.9 1012 10.11.12 1013 13.14.15 1020 2.4.6 1021 4.6.8 1022 6.8.10 1023 10.12.14\\n"

IOC:
epics>dbgf _FS1_CH01:RFA_D2137:FVSN_RD_ALL
DBR_STRING[20]: "1000" "1.2.3" "1010" "4.5.6" "1011" "7.8.9"
 "1012" "10.11.12" "1013" "13.14.15" "1020" "2.4.6"
 "1021" "4.6.8" "1022" "6.8.10" "1023" "10.12.14"

dirk-zimoch, 2018-12-14:
Hello Evan,

The problem you experienced has its origin in the way how %s and StreamDevice process input. Streamdevice processes input sequentially, that means while looping over the array it repetitively first parses the format and then checks for separators. It does not look ahead for separators before parsing the format. "%s" on the other hand reads characters until it finds a blank. Thus it reads the "," as well. To avoid this you can use a more specific format than "%s", for example "%[^,]" (read everything except commas).

I hope this makes things clearer.
Dirk

closed 2018-12-14 by dirk-zimoch

daykin, 2018-12-14:
That worked, thanks!