paulscherrerinstitute/StreamDevice #27: Release 2.8.8 does not build with Visual Studio 2010

MarkRivers, 2018-12-21:
When I try to build 2.8.8 with Visual Studio 2010 I get this error:
```
make[3]: Entering directory 'J:/epics/devel/stream/StreamDevice/src/O.windows-x64-vs2010'
cl -EHsc -GR               -nologo -FC -D__STDC__=0 -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE -DUSE_TYPED_RSET   -Ox -GL -Oy-   -W3 -w44355 -w44344 -w44251     -D_SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING    -MD -DEPICS_BUILD_DLL -DEPICS_CALL_DLL -TP  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/msvc -I../../../include/os/WIN32 -I../../../include      -IJ:/epics/devel/asyn-4-33/include   -IJ:/epics/devel/calc-3-7/include   -IJ:/epics/devel/sscan-2-11-1/include -IH:/epics-devel/base-7.0.2/include/compiler/msvc -IH:/epics-devel/base-7.0.2/include/os/WIN32 -IH:/epics-devel/base-7.0.2/include        -c ../DebugInterface.cc
DebugInterface.cc
j:\epics\devel\stream\streamdevice\src\streambusinterface.h(29) : error C2051: case expression not constant
j:\epics\devel\stream\streamdevice\src\streambusinterface.h(29) : warning C4065: switch statement contains 'default' but no 'case' labels
make[3]: *** [H:/epics-devel/base-7.0.2/configure/RULES_BUILD:239: DebugInterface.obj] Error 2
make[3]: Leaving directory 'J:/epics/devel/stream/StreamDevice/src/O.windows-x64-vs2010'
make[2]: *** [H:/epics-devel/base-7.0.2/configure/RULES_ARCHS:58: install.windows-x64-vs2010] Error 2
make[2]: Leaving directory 'J:/epics/devel/stream/StreamDevice/src'
make[1]: *** [H:/epics-devel/base-7.0.2/configure/RULES_DIRS:84: src.install] Error 2
make[1]: Leaving directory 'J:/epics/devel/stream/StreamDevice'
make: *** [configure/RULES_DIRS:88: StreamDevice.install] Error 2
```

It builds OK with Visual Studio 2015 and Visual Studio 2017.

kgofron, 2018-12-27:
I am getting the same error with Visual Studio 2010, Windows10_64bit, perl: v5.28.1, GNU Make 4.2.1;

1. Cloned stream from https://github.com/epics-modules/stream
2. Deleted existing folder StreamDevice and replaced with cloned StreamDevice repo from: https://github.com/paulscherrerinstitute/StreamDevice
3. Change tag within StreamDevice repo: 'git checkout 2.8.8'

make[3]: Entering directory 'c:/epics/support/stream/StreamDevice/src/O.windows-x64'

cl -EHsc -GR               -nologo -FC -D__STDC__=0 -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE -DUSE_TYPED_RSET   -Ox -GL -Oy-   -W3 -w44355 -w44344 -w44251     -D_SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING    -MD -DEPICS_BUILD_DLL -DEPICS_CALL_DLL -TP  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/msvc -I../../../include/os/WIN32 -I../../../include      -Ic:/epics/support/asyn/include   -Ic:/epics/support/calc/include   -Ic:/epics/support/sscan/include -Ic:/epics/base-7.0.2/include/compiler/msvc -Ic:/epics/base-7.0.2/include/os/WIN32 -Ic:/epics/base-7.0.2/include        -c ../DebugInterface.cc
DebugInterface.cc
c:\epics\support\stream\streamdevice\src\streambusinterface.h(29) : error C2051: case expression not constant
c:\epics\support\stream\streamdevice\src\streambusinterface.h(29) : warning C4065: switch statement contains 'default' but no 'case' labels
make[3]: *** [c:/epics/base-7.0.2/configure/RULES_BUILD:239: DebugInterface.obj] Error 2
make[3]: Leaving directory 'c:/epics/support/stream/StreamDevice/src/O.windows-x64'
make[2]: *** [c:/epics/base-7.0.2/configure/RULES_ARCHS:58: install.windows-x64] Error 2
make[2]: Leaving directory 'c:/epics/support/stream/StreamDevice/src'
make[1]: *** [c:/epics/base-7.0.2/configure/RULES_DIRS:84: src.install] Error 2
make[1]: Leaving directory 'c:/epics/support/stream/StreamDevice'
make: *** [configure/RULES_DIRS:88: StreamDevice.install] Error 2

dirk-zimoch, 2019-01-03:
It seems VS 2010 is not happy with my macro magic. It uses variadic macros. 
As it works fine with VS 2015, I consider this a bug in VS, not in StreamDevice.
I do not have VS 2010 available at the moment, so I cannot test anything.

MarkRivers, 2019-01-03:
Everything else in EPICS base (including 7.0.2), areaDetector, and synApps builds fine with VS2010.

dirk-zimoch, 2019-01-09:
Currently I cannot look into this because I do not have such an old Visual Studio Version available at the moment. Earliest I can have a look in February after my vacation.

FreddieAkeroyd, 2019-06-08:
The issue seems to be caused by the treatment of __VA_ARGS__ in Visual Studio 2010, see https://stackoverflow.com/questions/5134523/msvc-doesnt-expand-va-args-correctly for more details.

I have posted a potential fix at #40

dirk-zimoch, 2019-07-18:
Many thanks for the fixes, Freddy!
Nothing broke for my Linux targets, so I merged it.

closed 2019-07-18 by dirk-zimoch