paulscherrerinstitute/StreamDevice #33: Bad init_rec return value <PVname> ao: init_record

hinxx, 2019-04-12:
I have these two PVs:

```
# test1
record(ao, "$(P)$(R)SOUR:CURR:LIM1")
{
#    field(DTYP, "stream")
}

record(ao, "$(P)$(R)SOUR:CURR:LIM")
{
    field(DTYP, "stream")
}
```

After starting of the IOC I get the line:
```
Bad init_rec return value  PV: PBILAB:DC1:SOUR:CURR:LIM ao: init_record
```

I do not get the line for the first record, only for the second (DTYP=stream).

On 64-bit Linux, epics base 3.15.5, streamDevice 2.7.11, asyn R4-33.

dirk-zimoch, 2019-04-15:
Your first record does not have a DTYP. Thus it uses "Soft Channel". For "Soft Channel" an empty INP is totally fine, initializing the record with 0.
Your second record uses DTYP="stream" but does not define an INP link, which is required by the driver.The INP link should contain the used protocol file, the name of the protocol, optionally parameters to the protocol, and the IO port. As StreamDevice cannot find any of these in an empty INP, it returns an error.
Also in versions 2.7.7 up to 2.7.11 errors during initialization were disabled by default. Try 'var streamError 1' before 'iocInit' to see what exactly StreamDevice does not like. Earlier versions then 2.7.7 have error messages enabled by default. Later versions have error messages enabled during initialization and only switch them off by default afterwards.

hinxx, 2019-04-15:
> Your second record uses DTYP="stream" but does not define an INP link,

Should that be OUT link since it is ao record?

I've added this to the second record:
```
    field(OUT,  "@tldc2200.proto setF(SOUR:CURR:LIM) $(PORT) 0")
```

In any case I've added the suggested 'var streamError 1' and I can see the problem:

```
2019/04/15 14:18:23.162989 DET.DC1 PBILAB:DC1:SOUR:CURR:LIM: asynError in read. Asyn driver says: "Bulk read failed: Operation timed out"
2019/04/15 14:18:23.163196 DET.DC1 PBILAB:DC1:SOUR:CURR:LIM: I/O error after reading 0 bytes: ""
2019/04/15 14:18:23.163310 _main_ PBILAB:DC1:SOUR:CURR:LIM: @init handler failed
2019/04/15 14:18:23.163370 _main_ PBILAB:DC1:SOUR:CURR:LIM: Record initialization failed
Bad init_rec return value  PV: PBILAB:DC1:SOUR:CURR:LIM ao: init_record
```

I seem to be getting this error on IOC startup at semi-random occasions. Also, it does not really cause any trouble to the operation after the IOC has started (I can still operate the PV in question and does what is expected).

I think I'm looking at hardware trouble here. Thank you for your time @dirk-zimoch !

dirk-zimoch, 2019-04-15:
> Should that be OUT link since it is ao record?
Yes.

> I think I'm looking at hardware trouble here.
May or may not be. Have you played with the timeouts in StreamDevice, in particular ReplyTimeout and ReadTimeout? StreamDevices passes them to the asyn layer. Here your ReplyTimeout seems to be too short. Or the device really has problems.

hinxx, 2019-04-15:
I have not touched any of the timeout settings. It’s an USB TMC device and I use usbtmcConfigure() in startup file. I think I noticed some Ni Visa code/samples that came with it, I’ll check to see if any special handling is made on that part.

dirk-zimoch, 2019-04-29:
Not a problem in StreamDevice. Message comes from drvAsynUSBTMC having problems reading from the device.

closed 2019-04-29 by dirk-zimoch