paulscherrerinstitute/StreamDevice #42: device support should set a non-zero status when startProtocol fails

bfrk, 2019-09-10:
`bool Stream::process()` initializes the status to `NO_ALARM`, then calls `startProtocol`, and if that fails sets `recGblSetSevr(record, status, INVALID_ALARM)`. This means the record will have `STAT=NO_ALARM` and `SEVR=INVALID` which is a bit confusing.

Here is a fix, using the `COMM_ALARM` status instead.

```
diff --git a/src/StreamEpics.cc b/src/StreamEpics.cc
index 9117fd9..1f16e2d 100644
--- a/src/StreamEpics.cc
+++ b/src/StreamEpics.cc
@@ -732,7 +732,7 @@ process()
         return false;
     }
     debug("Stream::process(%s) start\n", name());
-    status = NO_ALARM;
+    status = COMM_ALARM;
     convert = OK;
     record->pact = true;
     if (!startProtocol(StreamCore::StartNormal))

```

chrschroeder, 2019-09-10:
A short comment: this will be a problem on disconnects, which can easily be simulated by setting of autoconnect and disconnect through CNCT field of the asyn record.

dirk-zimoch, 2019-09-11:
You are right, NO_ALARM/INVALID is a confusing combination and was not intended.
But I need to check your fix with "I/O Intr" records first.