paulscherrerinstitute/StreamDevice #49: testTimeStamp question

LeeHudsonDLS, 2020-02-24:
Hi,
I'm trying to get the tests in this module working and i'm stuck on testTimeStamp. Can someone tell me anything about the "Epics" package that's required by this test? Is it a TCL package? If so do you know where I can get it from?

dirk-zimoch, 2020-03-02:
It is a tcl package that we have at PSI. It is based on this: https://epics.anl.gov/EpicsDocumentation/ExtensionsManuals/TclTk/et.tcltk.html
As I wrote the tests for me, I did not spent any effort to make them run everywhere. Thea are from a time when this repo was not yet publicly available.

dirk-zimoch, 2020-03-02:
See "caTCL" here: https://epics.anl.gov/extensions/tcl/index.php

LeeHudsonDLS, 2020-03-11:
Ok thanks for the clarification

closed 2020-03-11 by LeeHudsonDLS