paulscherrerinstitute/StreamDevice #50: Brackets in parameter no longer work

LeeHudsonDLS, 2020-02-25:
Hi,
I'm in the process of updating our version of streamDevice to your latest release. I have a module that no longer works. The IOC seems to be unhappy about the following record field:

record(ai, "BL99I-LCH-LPP-01:STREAM_BREAKER")
{
    field(INP,  "@laserPuckPointer.proto readVariable(M_SSPD(10)) LPORT")
}

The bit in particular appears to be "M_SSPD(10)". It doesn't like the  brackets in the parameter:

 _main_ Cannot find a bus named ')' for 'BL99I-LCH-LPP-01:STREAM_BREAKER'
 
Do I need to add some kind of escape characters? The method definition in the protocol file looks like this:

readVariable { out "1;9;VAL\$1" ; in "Qo%*c\$1=%f" ; }

I would expect the following string to get written to the device when the record is processed:

1;9;VALM_SSPD(10)

This used to work with streamDevice 2_5.
Thanks

dirk-zimoch, 2020-02-25:
Thanks for your bug report.
I had changed parameter parsing to allow spaces around commas in the list. So I am now looking for the closing ). It seems that gets confused when ) is actually part of the parameter string. I need to fix that. Unfortunately I have no quick solution for you at the moment.

LeeHudsonDLS, 2020-02-26:
Ok thanks for the clarification. I wonder if it would be possible to keep track of how many brackets are open. If you would like to point me in the right direction of this logic i'm happy to take a look at a fix. I'm not overly familiar with the internal workings of this module so i'm not sure where I should look for this. Thanks for the swift response!

dirk-zimoch, 2020-02-26:
At the moment, I simply use `sscanf(...,"%[^)]",...)`, in other words fetch everything until the next `)`.
To track open brackets requires some major redesign and opens a whole new can of worms: How to parse for example `xxx((1,2),3)`? Has it 2 arguments, `(1,2)` and `3` or 3 arguments `(1`, `2)`, and `3`? So commas within embedded brackets are now an issue. How to pass a single opening `(` without a corresponding `)` as an argument? Escaped brackets and commas (and escape chars)?
I need a few days to work out something...

LeeHudsonDLS, 2020-02-26:
I understand. Let me know when you want me to do more testing.
Thanks

dirk-zimoch, 2020-02-27:
I have pushed a fix that should do the trick for you. It does not yet handle embedded commas or non-balanced `()` yet.

LeeHudsonDLS, 2020-02-27:
Perfect thanks, i'll give it a try.

LeeHudsonDLS, 2020-02-27:
Thanks for the fix, that appear to work. Thanks again

closed 2020-02-27 by LeeHudsonDLS