paulscherrerinstitute/StreamDevice #55: %d, %X, %04X result in error

bsobhani, 2020-04-30:
I have a record that used to work fine, but when I updated StreamDevice I now get "Cannot format value with '%04X'" when I write to it. 

The full entry from the protocol file is: 

WritePID {
        out "*\$1W\$2%04X";
        out "*\$1Z02";
        wait 5000;
}

The record type was originally ao, so I tried changing the type to longout, but the problem persists. Right now the record looks like:

record(longout, "$(Sys)$(Dev)Gain:PB$(N_GAIN)-SP")
{
        field(DTYP, "stream")
        field(DESC, "Proportional Band 1")
        field(OUT, "@omega-CNi32.proto WritePID($(Chan),17) $(PORT)")
        field(EGU,  "Counts")
        field(PINI, "YES")
        field(VAL, "20")

        info(autosaveFields_pass0, "VAL")
}

The only thing I can do is to format the get the message to go away is changing the %04X to %lf. I also tried %d and %d does not work. Any ideas for why %lf works but not %04X or %d?

dirk-zimoch, 2020-04-30:
Are you using EPICS 7? I think I messes it up. May be related to pupp request 52.

bsobhani, 2020-04-30:
Yes, the EPICS version is:
## EPICS R7.0.3.2-DEV
## Rev. R7.0.3.1-102-g1d6fcd46d653934da0e2

The StreamDevice versions is:   StreamDevice 2.8.12 built Apr 29 2020 21:20:09

dirk-zimoch, 2020-05-04:
Today's commit cede631 should fix this problem. But I am still running tests....

bsobhani, 2020-05-04:
I pulled the latest changes and I confirm this seems to have resolved the issue. Thank you.

closed 2020-05-04 by bsobhani