paulscherrerinstitute/StreamDevice #56: compile error from StreamVersion.c

prjemian, 2020-06-07:
I'm updating my docker image for synApps with EPICS base 7.0.4 and am 
running into this compiler error.  Do you know what can be the problem?

```
root@006b1d9d277b:/opt/synApps/support/StreamDevice-2-8-14# make
If you are not using the PSI build environment, GNUmakefile can be removed.
make -C ./src install
make[1]: Entering directory '/opt/synApps/support/StreamDevice-2-8-14/src'
make -C O.linux-x86_64 -f ../Makefile TOP=../../.. \
     T_A=linux-x86_64 install
make[2]: Entering directory 
'/opt/synApps/support/StreamDevice-2-8-14/src/O.linux-x86_64'
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_ 
-DUNIX  -Dlinux    -DUSE_TYPED_RSET -DSTREAM_INTERNAL   -O3   -Wall 
-mtune=generic      -m64 -fPIC -I. -I../O.Common -I. -I. -I.. 
-I../../../include/compiler/gcc -I../../../include/os/Linux 
-I../../../include   -I/opt/synApps/support/include 
-I/opt/base/include/compiler/gcc -I/opt/base/include/os/Linux 
-I/opt/base/include   -I/opt/synApps/support/alive-master/include 
-I/opt/synApps/support/asyn-master/include 
-I/opt/synApps/support/autosave-master/include/os/Linux 
-I/opt/synApps/support/autosave-master/include 
-I/opt/synApps/support/busy-master/include 
-I/opt/synApps/support/calc-master/include 
-I/opt/synApps/support/iocStats-master/include/os/Linux 
-I/opt/synApps/support/iocStats-master/include 
-I/opt/synApps/support/ipac-master/include 
-I/opt/synApps/support/lua-master/include 
-I/opt/synApps/support/modbus-master/include 
-I/opt/synApps/support/motor-master/include 
-I/opt/synApps/support/optics-master/include 
-I/opt/synApps/support/sscan-master/include 
-I/opt/synApps/support/std-master/include 
-I/opt/synApps/support/areaDetector-master/ADCore/include 
-I/opt/synApps/support/areaDetector-master/ADSupport/include/os/Linux 
-I/opt/synApps/support/areaDetector-master/ADSupport/include 
-I/opt/synApps/support/seq-2-2-6/include        -c ../StreamVersion.c
../StreamVersion.c:31:5: error: expected ',' or ';' before 'STREAM_DEV'
    31 |     STREAM_DEV
       |     ^~~~~~~~~~
make[2]: *** [/opt/base/configure/RULES_BUILD:241: StreamVersion.o] Error 1
make[2]: Leaving directory 
'/opt/synApps/support/StreamDevice-2-8-14/src/O.linux-x86_64'
make[1]: *** [/opt/base/configure/RULES_ARCHS:58: install.linux-x86_64] 
Error 2
make[1]: Leaving directory '/opt/synApps/support/StreamDevice-2-8-14/src'
make: *** [/opt/base/configure/RULES_DIRS:85: src.install] Error 2
root@006b1d9d277b:/opt/synApps/support/StreamDevice-2-8-14#
```

The build is in Ubuntu 20.04 OS, 64-bit.

version | compiles?
--- | ---
master | fails
2.1.14 | fails
2.1.12 | compiles

dirk-zimoch, 2020-06-08:
What does the generated file O.Common/StreamVersion.h look like?
Mine is:
`/* Generated file StreamVersion.h */

#ifndef StreamVersion_h
#define StreamVersion_h

#define STREAM_MAJOR 2
#define STREAM_MINOR 8
#define STREAM_PATCHLEVEL 14
#define STREAM_DEV ""

#endif /* StreamVersion_h */
`

GabeYcas2, 2020-06-19:
I am also having this problem. It looks like the StreamVersion.h file is being generated by a Perl script, which relies upon version information from the StreamDevice Git repo (ie the .git directory.) When downloading a zipped copy of the source code (I used the release), this script fails and generates an empty StreamVersion.h file. This then causes the build error.

Perhaps a fix could be made by altering the makeStreamVersion.pl to use an alternate source for versions if the git command fails (currently, the failure results in 'die "Cannot run git.\n";')

dirk-zimoch, 2020-06-19:
Does anyone know how to generate the file on the git server and include it in the zip file?

dirk-zimoch, 2020-06-30:
For the moment, do not download the zip file. Clone the repo.

closed 2020-06-30 by dirk-zimoch

dirk-zimoch, 2020-08-18:
Should be solved now in 2.8.16