paulscherrerinstitute/StreamDevice #57: StreamDevice 2.18.14 failed to compile with EPICS 7.0.4 on windows x64

LeeYangLBLBCS, 2020-07-14:
2.8.13 compiled fine, but 2.18.14 gives me the following error: (windows 10 x64):
cl               -nologo -FC -D__STDC__=0 -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DUSE_TYPED_RSET -DSTREAM_INTERNAL   -Ox -GL -Oy-   -W3        -MD -DEPICS_BUILD_DLL -DEPICS_CALL_DLL   -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/msvc -I../../../include/os/WIN32 -I../../../include   -IC:\epics\modules\synApps_6_1_epics7\support/include -IC:\epics\base-7.0.4/include/compiler/msvc -IC:\epics\base-7.0.4/include/os/WIN32 -IC:\epics\base-7.0.4/include   -IC:\epics\modules\synApps_6_1_epics7\support/alive-R1-1-1/include   -IC:\epics\modules\synApps_6_1_epics7\support/asyn-R4-38/include  -IC:\epics\modules\synApps_6_1_epics7\support/autosave-R5-10/include/os/WIN32 -IC:\epics\modules\synApps_6_1_epics7\support/autosave-R5-10/include   -IC:\epics\modules\synApps_6_1_epics7\support/busy-R1-7-2/include   -IC:\epics\modules\synApps_6_1_epics7\support/calc-R3-7-4/include   -IC:\epics\modules\synApps_6_1_epics7\support/camac-R2-7-1/include            -IC:\epics\modules\synApps_6_1_epics7\support/dxp-R6-0/include   -IC:\epics\modules\synApps_6_1_epics7\support/dxpSITORO-R1-2/include  -IC:\epics\modules\synApps_6_1_epics7\support/iocStats-3-1-16/include/os/WIN32 -IC:\epics\modules\synApps_6_1_epics7\support/iocStats-3-1-16/include      -IC:\epics\modules\synApps_6_1_epics7\support/ipac-2-15/include   -IC:\epics\modules\synApps_6_1_epics7\support/ip330-R2-9/include         -IC:\epics\modules\synApps_6_1_epics7\support/lua-R2-0/include  -IC:\epics\modules\synApps_6_1_epics7\support/mca-R7-8/include/os/WIN32 -IC:\epics\modules\synApps_6_1_epics7\support/mca-R7-8/include   -IC:\epics\modules\synApps_6_1_epics7\support/measComp-R2-3/include   -IC:\epics\modules\synApps_6_1_epics7\support/modbus-R3-0/include   -IC:\epics\modules\synApps_6_1_epics7\support/motor-R7-2-1/include   -IC:\epics\modules\synApps_6_1_epics7\support/optics-R2-13-3/include   -IC:\epics\modules\synApps_6_1_epics7\support/quadEM-R9-2-1/include   -IC:\epics\modules\synApps_6_1_epics7\support/softGlue-R2-8-2/include   -IC:\epics\modules\synApps_6_1_epics7\support/softGlueZynq-R2-0-2/include   -IC:\epics\modules\synApps_6_1_epics7\support/sscan-R2-11-3/include   -IC:\epics\modules\synApps_6_1_epics7\support/std-R3-6-1/include      -IC:\epics\modules\synApps_6_1_epics7\support/vac-R1-9/include   -IC:\epics\modules\synApps_6_1_epics7\support/vme-R2-9-2/include            -IC:\epics\modules\synApps_6_1_epics7\support/areaDetector-R3-7/ADCore/include  -IC:\epics\modules\synApps_6_1_epics7\support/areaDetector-R3-7/ADSupport/include/os/WIN32 -IC:\epics\modules\synApps_6_1_epics7\support/areaDetector-R3-7/ADSupport/include   -IC:\epics\modules\synApps_6_1_epics7\support/areaDetector-R3-7/ADSimDetector/include   -IC:\epics\modules\synApps_6_1_epics7\support/seq-2-2-6/include   -IC:\epics\modules\synApps_6_1_epics7\support/allenBradley-2-3/include        -c ../StreamVersion.c
StreamVersion.c
C:\epics\modules\synApps_6_1_epics7\support\StreamDevice-2-8-14\src\StreamVersion.c(31): error C2146: syntax error: missing ';' before identifier 'STREAM_DEV'
C:\epics\modules\synApps_6_1_epics7\support\StreamDevice-2-8-14\src\StreamVersion.c(32): error C2143: syntax error: missing '{' before 'string'
C:\epics\modules\synApps_6_1_epics7\support\StreamDevice-2-8-14\src\StreamVersion.c(32): error C2059: syntax error: 'string'
gmake[2]: *** [C:\epics\base-7.0.4/configure/RULES_BUILD:241: StreamVersion.obj] Error 2
gmake[2]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-14/src/O.windows-x64'
gmake[1]: *** [C:\epics\base-7.0.4/configure/RULES_ARCHS:58: install.windows-x64] Error 2
gmake[1]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-14/src'
gmake: *** [C:\epics\base-7.0.4/configure/RULES_DIRS:85: src.install] Error 2

dirk-zimoch, 2020-07-15:
Duplicate of #56

MarkRivers, 2020-07-21:
This can be fixed by installing Git on the Windows machine.

I first ran into this problem when building base EPICS base 7.0.4.  For that it is sufficient to create a git.bat file in your PATH that contains a single line like this:
@echo GSECARS-Build

That will cause the "git" command to succeed and return a string.  You don't actually need git to be installed.

But for StreamDevice that trick does not work, because it actually parses the string that is returned, and fails for a string like GSECARS-Build.

dirk-zimoch, 2020-07-22:
i am working on a solution based on .gitattributes and a version file that get filled in with the necessary information when github builds the zip file...

MarkRivers, 2020-07-22:
> i am working on a solution based on .gitattributes and a version file that get filled in with the necessary information when github builds the zip file...

That would not help installations like mine.  My working directory is a git clone on a Linux file server.  I build for all ARCHs in that directory (various Linux versions, Windows, vxWorks).  It works OK if I build on Linux first (has git) and then Windows (no git).  But if I do a "make clean; make" on Windows then the build fails because it needs to regenerate the version file.

dirk-zimoch, 2020-08-18:
Should be solved now in 2.8.16.

closed 2020-08-18 by dirk-zimoch