paulscherrerinstitute/StreamDevice #58: STREAM_PROTOCOL_PATH is incorrectly parsed on Windows

MarkRivers, 2020-07-21:
StreamProtocol.cc contains this code:
```
#ifdef windows
     const char pathseparator = ';';
     const char dirseparator = '\\';
 #else
```

The problem is that "windows" is not defined when building the code, at least not with Visual Studio 2017 and base 7.0.4.  This means that it is using the incorrect pathseparator and dirseparator. 
 
This tech-talk issue https://epics.anl.gov/tech-talk/2020/msg01400.php shows how in spite of this it actually works in some cases, which may be why the problem has not been previously reported.

This patch fixes the problem.
```
diff --git a/src/StreamProtocol.cc b/src/StreamProtocol.cc
index 77741b9..a60b6b0 100644
--- a/src/StreamProtocol.cc
+++ b/src/StreamProtocol.cc
@@ -157,7 +157,7 @@ StreamProtocolParser* StreamProtocolParser::
 readFile(const char* filename)
 {
     FILE* file;
-#ifdef windows
+#ifdef _WIN32
     const char pathseparator = ';';
     const char dirseparator = '\\';
 #else
```

dirk-zimoch, 2020-07-22:
Oops.

dirk-zimoch, 2020-08-18:
Fixed

closed 2020-08-18 by dirk-zimoch