paulscherrerinstitute/StreamDevice #59: stream is installing into $(TOP)/../lib rather than into $(TOP)/lib

MarkRivers, 2020-07-21:
This problem is described in this tech-talk thread: https://epics.anl.gov/tech-talk/2020/msg01372.php
 
I am running stream 2.8.14.  I am having a problem that when I build it, the files get installed into $(TOP)/../lib, not into $(TOP)/lib.

This is the git diff of my files and the ones in 2.8.14.  I have only commented out streamApp in the top-level Makefile, and modified configure/RELEASE for new locations.
```
corvette:~/devel/stream>git diff -w -b 2.8.14
diff --git a/Makefile b/Makefile
index 4e56eec..1312552 100644
--- a/Makefile
+++ b/Makefile
@@ -38,7 +38,7 @@ else
endif
DIRS += src
-DIRS += streamApp
+#DIRS += streamApp
streamApp_DEPEND_DIRS = src
include $(CONFIG)/RULES_TOP

diff --git a/configure/RELEASE b/configure/RELEASE
index 72efa75..5b80a63 100644
--- a/configure/RELEASE
+++ b/configure/RELEASE
@@ -14,15 +14,15 @@ TEMPLATE_TOP=$(EPICS_BASE)/templates/makeBaseApp/top
# define INSTALL_LOCATION_APP here
#INSTALL_LOCATION_APP=<fullpathname>

-SUPPORT=$(TOP)/..
+SUPPORT=/corvette/home/epics/devel
-include $(TOP)/../configure/SUPPORT.$(EPICS_HOST_ARCH)
-ASYN=$(SUPPORT)/asyn4-36
-CALC=$(SUPPORT)/calc-3-7
+ASYN=$(SUPPORT)/asyn-4-40
+CALC=$(SUPPORT)/calc-3-7-3
PCRE=$(SUPPORT)/pcre-7-2

 # EPICS_BASE usually appears last so other apps can override stuff:
-EPICS_BASE=/usr/local/epics/base-7.0.3
+EPICS_BASE=/corvette/usr/local/epics-devel/base-7.0.4

# These lines allow developers to override these RELEASE settings
# without having to modify this file directly.
```

I do a realclean realuninstall and do an ls in stream/ and in stream/../lib.  There are no lib/ directories.

```
corvette:~/devel/stream>make -sj realclean realuninstall
If you are not using the PSI build environment, GNUmakefile can be removed.

corvette:~/devel/stream>ls
GNUmakefile  LICENSE  LICENSE.LESSER  Makefile  README.md  config  configure  docs  src  streamApp

corvette:~/devel/stream>ls ../lib
ls: cannot access ../lib: No such file or directory
```

I then do the build.  After the build there is no stream/lib directory, but there is a stream/../lib/linux-x86_64 directory containing the stream libraries.
```
corvette:~/devel/stream>make -sj
If you are not using the PSI build environment, GNUmakefile can be removed.
Creating ../O.Common/StreamVersion.h from git tag

corvette:~/devel/stream>ls
GNUmakefile  LICENSE  LICENSE.LESSER  Makefile  README.md  config  configure  docs  src  streamApp

corvette:~/devel/stream>ls ../lib
linux-x86_64

corvette:~/devel/stream>ls ../lib/linux-x86_64/
libstream.a  libstream.so
```
 If I search for INSTALL_LOCATION in stream/configure I don’t see the problem.

 ```
corvette:~/devel/stream/configure>grep INSTALL_LOCATION *
CONFIG_APP:INSTALL_LOCATION = $(TOP)
CONFIG_APP:ifdef INSTALL_LOCATION_APP
CONFIG_APP:INSTALL_LOCATION = $(INSTALL_LOCATION_APP)
grep: O.Common: Is a directory
grep: O.linux-x86_64: Is a directory
RELEASE:# define INSTALL_LOCATION_APP here
RELEASE:#INSTALL_LOCATION_APP=<fullpathname>
```

Am I doing something wrong, or why does the distribution version of stream install into $(TOP)/..?

dirk-zimoch, 2020-07-22:
This was more or less on purpose to be backward compatible to the time when Stream did not have its own top.
If it finds ../configure, it uses that one and installs there. Itherwise it uses its own configure and installs into its own dir.

MarkRivers, 2020-07-22:
> This was more or less on purpose to be backward compatible to the time when Stream did not have its own top.
If it finds ../configure, it uses that one and installs there. Itherwise it uses its own configure and installs into its own dir.

Many sites put stream under the top-level synApps directory, which does have a configure.  stream is the only synApps module which installs into $(TOP)/../

I would like to change stream to behave like the other modules, so I don't have those directories in $(TOP)/../ that contain only stream files.   What is causing it to behave differently from other synApps modules like asyn?  I could not find anything in stream/configure that was doing that.  How do I disable this behavior?

dirk-zimoch, 2020-08-18:
Should be solved now in 2.8.16.

closed 2020-08-18 by dirk-zimoch