paulscherrerinstitute/StreamDevice #60: Add a new checksum method

ljsSally, 2020-08-13:
StreamDevice has many checksum functions, but i have some devices which need LRC checksum, so i write a function about LRC checksum.
example: 
:010310040002E6      here E6 is LRC checksum. Calculation process: 01+03+10+04+00+02=1A(hex), negative of sum is E5, then add 1 is E6.
/////*add a new checksum function*/////
//This function is to implement LRC verification
//static unsigned int lrcsum(const unsigned char* data,unsigned int len,unsigned int sum)
static ulong lrcsum(const uchar* data,ulong len,ulong sum)
{
	//Add all hex digits,ignore all other bytes.
	unsigned int d;
	while(len--)
	{
		d=toupper(*data++);
		if(isxdigit(d))
		{
			if(isdigit(d))
				d -= '0';
			else 
				d -= 'A'-0x0A;
			if(len%2)
				d *= 16;
			sum += d;		
		}
	}
	return sum;	
}
static checksum checksumMap[] =
// You may add your own checksum functions to this map.
{
//    name      func              init        xorout  bytes  chk("123456789")
    {"sum",     sum,              0x00,       0x00,       1}, // 0xDD
    {"sum8",    sum,              0x00,       0x00,       1}, // 0xDD
...
    {"adler32", adler32,          0x00000001, 0x00000000, 4}, // 0x091E01DE
    {"hexsum8", hexsum,           0x00,       0x00,       1},  // 0x2D
    {"lrcsum8", lrcsum,           0x00,       0x00,       1},//add lrcsum8
    {"-lrcsum8",lrcsum,           0xFF,       0xFF,       1}//add -lrcsum8
};

dirk-zimoch, 2020-08-13:
Is it this one? https://en.wikipedia.org/wiki/Longitudinal_redundancy_check
I will add your implmentation to the next release.

dirk-zimoch, 2020-08-13:
Looks different. What does the name LRC mean if it is not Longitudinal Redundancy Check?

ljsSally, 2020-08-13:
Yes,LRC is Longitudinal Redundancy Check.

ljsSally, 2020-08-13:
> Is it this one? https://en.wikipedia.org/wiki/Longitudinal_redundancy_check
> I will add your implmentation to the next release.

Yes.

dirk-zimoch, 2020-08-18:
Not in this release, but I will have a look at it. First I need to understand your implementation.
It does not do the checksum over all the bytes like the original LRC does, but over the hexadecimal ASCII representation of bytes (like my hexsum does). Thus I will probably give it another name and reserve "lrc" for the implementation following the wikipedia article.

marciodo, 2021-03-11:
I'm working on this issue. Hopefully will finish until tomorrow.

dirk-zimoch, 2021-03-12:
If I understand this correctly, the checksum does not work on the text bytes themselves (like most other checksums) but on their interpretation as hex numbers in ASCII text. The checksum function needs to consume 2 chars at each step, convert them to a 1 byte hex number and work on that one.

ljsSally, 2021-03-15:
Yes, I agree with you. Here are some equipment checksums that need to be handled in this way.

marciodo, 2021-03-15:
@ljsSally, I think you were going to paste some checksum examples from your equipment, but they were missing on the post. If you have them it would be nice.

marciodo, 2021-03-15:
> If I understand this correctly, the checksum does not work on the text bytes themselves (like most other checksums) but on their interpretation as hex numbers in ASCII text. The checksum function needs to consume 2 chars at each step, convert them to a 1 byte hex number and work on that one.

Thanks, @dirk-zimoch. This is the kind of clarification that I was expecting. So, I already have an LRC function that works like all the others, operating on the ASCII byte codes. Now I need to create another one interpreting the ASCII as hex numbers. I was not sure what the default behavior should be for Stream Device.

ljsSally, 2021-03-22:
My equipment checksums and command, for example:
Send:":0117010000050109000000D8"   
"D8" is the LRC checksum, and checksum calulation: 01+17+01+05+01+09=28(Hex), NOT(28)+1=D8 
Recive:":01170CC1005EF40000000000000000C9"
“C9” is the LRC checksum，and checksum calculation: 01+17+0C+C1+5E+F4=237(Hex), NOT(237)+1=C9

dirk-zimoch, 2021-07-05:
Included in 2.8.20.

closed 2021-07-05 by dirk-zimoch