paulscherrerinstitute/StreamDevice #62: 2.8.16 not on branch

LeeHudsonDLS, 2020-12-10:
Hi,
I would like to merge the latest release onto our master branch but I see that this tag wasn't made on a branch that currently exists (or perhaps it's a local one). Was this intentional?
Thanks
Lee

dirk-zimoch, 2020-12-10:
When I had started on 2.8, i had used a branch "2-8" and had 2.7 on "master".
Later I renamed the old "master" to "2-7" and  "2-8" to "master"
Only later I noticed that this is a problem when you had the old "master" cheked out.
Sorry, I am no git expert. I annot tell you how to solve that problem.

LeeHudsonDLS, 2020-12-10:
OK no worries, i'm not sure what the solution is either apart from me merging that commit directly. Even when I do a fresh clone it's still showing 2.6.16 as not being on a branch. I don't know if you can checkout that commit locally, merge into master then push?

dirk-zimoch, 2020-12-10:
I pushed again. Maybe I did something wrong last time. My git showd me that only the commit before 2.8.16 had been pushed. I wonder how then the tag 2.8.16 could have been there. Anyway, now 2.8.17 is pushed.

LeeHudsonDLS, 2020-12-10:
Ah thanks that seems to have gotten it, thanks!

closed 2020-12-10 by LeeHudsonDLS