paulscherrerinstitute/StreamDevice #70: Release Notes?

anjohnson, 2021-03-22:
I don't see anything obvious in the documentation for StreamDevice that looks like a set of Release Notes, did I just miss them?

We currently have 2.8.8 installed. How can I find out what's changed between that and the latest release?

dirk-zimoch, 2021-03-23:
That is why `git log` exists, I suppose.
```
$ git log 2.8.7...2.8.8
commit acf7efcff2e29a997a5c2d21629d7ef64f1e19de
Author: Dirk Zimoch <dirk.zimoch@psi.ch>
Date:   Tue Nov 27 13:45:27 2018 +0100

    fix hanging 'in' command when it is the first command of the protocol and the device is offline

commit d873d220dc1c395d16eb0ec790832040941d4cf3
Author: Dirk Zimoch <dirk.zimoch@psi.ch>
Date:   Tue Nov 27 09:36:53 2018 +0100

    fix broken version detection
```

dirk-zimoch, 2021-07-05:
CHANGELOG.md added in release 2.8.20.

closed 2021-07-05 by dirk-zimoch

anjohnson, 2021-07-06:
Thanks, very useful.