paulscherrerinstitute/StreamDevice #72: An error occurred while installing StreamDevice

Lytt-2020, 2021-03-31:
System: CentOS7--Linux
When installing StreamDevice, make an error when compiling:
/usr/bin/g++ -D_GNU_SOURCE -D_DEFAULT_SOURCE -D_X86_64_ -DUNIX -Dlinux -DUSE_TYPED_RSET -DSTREAM_INTERNAL -O3 -g -Wall -mtune=generic -m64 -fPIC -I. -I../O.Common -I. -I . -I.. -I../../include/compiler/gcc -I../../include/os/Linux -I../../include -I/home/lytt/local/epics /base/include/compiler/gcc -I/home/lytt/local/epics/base/include/os/Linux -I/home/lytt/local/epics/base/include -c ../AsynDriverInterface.cc
../AsynDriverInterface.cc:41:24: Fatal error: asynDriver.h: No such file or directory
  #include "asynDriver.h"
Excuse me, where is the problem? How to solve it? Please guide me! Thank you!

Lytt-2020, 2021-03-31:
> I see no -I option with a path to the "asyn" module. Have you already built "asyn"?
>  Have you set the ASYN variable in configure/RELEASE to the correct installation path of the "asyn" module?
>  It seems you have set it to something, otherwise AsynDriverInterface.cc would not be built.

Hello! Thank you for your reply. I have installed "asyn", and I have set the installation path of the "asyn" module. I will paste the setting of "RELEASE" to the back and see if the format of the setting is correct. How to solve it? Thank you!
The contents of RELEASE are as follows:

TEMPLATE_TOP=$(EPICS_BASE)/templates/makeBaseApp/top

# If you don't want to install into $(TOP) then
# define INSTALL_LOCATION_APP here
#INSTALL_LOCATION_APP=<fullpathname&gt;

SUPPORT=$(TOP)/..
-include $(TOP)/../configure/SUPPORT.$(EPICS_HOST_ARCH)

ASYN=$(SUPPORT)/home/lytt/local/epics/asyn4-30
#CALC=$(SUPPORT)/calc-3-7
#PCRE=$(SUPPORT)/pcre-7-2

# EPICS_BASE usually appears last so other apps can override stuff:
EPICS_BASE=/home/lytt/local/epics/base

# These lines allow developers to override these RELEASE settings
# without having to modify this file directly.
-include $(TOP)/../RELEASE.local
-include $(TOP)/../RELEASE.$(EPICS_HOST_ARCH).local
-include $(TOP)/configure/RELEASE.local

dirk-zimoch, 2021-04-01:
After setting ASYN, have run run `make` in the top directory (or in configure)? Running `make` only in the src directory after changing the configuration is not sufficient. It is important that `make` runs in configure to set up the search path.

Lytt-2020, 2021-04-01:
Thank you for your reply! I ran make in the top directory, but the same error still occurred, and I have configured my installation path. The file asynDreiver.h can be found in asyn. May I ask where the environment configuration is wrong? I put the configuration content of RELEASE at the end, you can take a look. Thank you!
Contents of RELEASE:
#RELEASE Location of external products
# Run "gnumake clean uninstall install" in the application
# top directory each time this file is changed.
#
# NOTE: The build does not check dependencies on files
# external to this application. Thus you should run
# "gnumake clean uninstall install" in the top directory
# each time EPICS_BASE, SNCSEQ, or any other external
# module defined in the RELEASE file is rebuilt.

TEMPLATE_TOP=$(EPICS_BASE)/templates/makeBaseApp/top

# If you don't want to install into $(TOP) then
# define INSTALL_LOCATION_APP here
#INSTALL_LOCATION_APP=<fullpathname&gt;

SUPPORT=$(TOP)/..
-include $(TOP)/../configure/SUPPORT.$(EPICS_HOST_ARCH)

ASYN=$(SUPPORT)/home/lytt/epics/asyn4/asyn4-38
#CALC=$(SUPPORT)/home/lytt/epics/calc/calc_R3-4/calc-3-4
PCRE=$(SUPPORT)/home/lytt/epics/streamdevice/StreamDevice-master/prce/prce2-10.36

# EPICS_BASE usually appears last so other apps can override stuff:
EPICS_BASE=/home/lytt/epics/base

# These lines allow developers to override these RELEASE settings
# without having to modify this file directly.
-include $(TOP)/../RELEASE.local
-include $(TOP)/../RELEASE.$(EPICS_HOST_ARCH).local
-include $(TOP)/configure/RELEASE.local

dirk-zimoch, 2021-04-01:
Try to set ASYN to the absolute path:
`ASYN=/home/lytt/local/epics/asyn4-30`

Lytt-2020, 2021-04-01:
Thank you for your reply! According to your statement, I tried again and it still didn't work. Excuse me, are other file environments not set correctly? Thank you!

dirk-zimoch, 2021-04-01:
Where are your components installed? Sometimes you use the path `/home/lytt/local/epics/asyn4-30` but sometimes `/home/lytt/epics/asyn4/asyn4-38`
ASYN should be set to the parent directory of the include and lib directories, where asyn has installed its files.
Where is asynDriver.h? If it is /home/lytt/local/epics/asyn4-30/include/asynDriver.h, then /home/lytt/local/epics/asyn4-30 is the correct value.

Lytt-2020, 2021-04-01:
I'm very sorry, it took so long to get back to you! First of all thank you for your reply! Then the installation path of my component asyn is /home/lytt/epics/asyn4/asyn4-38. Among them, asyn4-38 contains include and lib directories. The path of my asynDriver.h is /asyn4-38/asyn/asynDriver/asynDriver.h, not in /include/asynDriver.h, is there something wrong with my file? I reinstalled it with CentOS7, but this problem still occurs. Is the version of asyn I use too high and incompatible with StreamDevice? Thank you very much for your attention!

dirk-zimoch, 2021-04-01:
Can it be that you have not yet built asyn? Go to your asyn4-38 and run make.

Lytt-2020, 2021-04-01:
I have re-compiled asyn4-38 according to what you said, and there is no error. How can I tell that asyn has been installed successfully, can it be done without reporting an error? Thank you!

dirk-zimoch, 2021-04-01:
Asyn should create directories `lib`, `include`, `dbd`  and a few others. If you have not set `INSTALL_LOCATION` it should have created those inside its own top location (asyn4-38). Do you find include/asynDriver.h there?

Lytt-2020, 2021-04-01:
I can find include/asynDriver.h. But I can’t find INSTALL_LOCATION. If I want to create it, how do I need to create it? Thank you!

dirk-zimoch, 2021-04-01:
Forget INSTALL_LOCATION. 
Set ASYN in StreamDevice to the parent directory of the include directory where asynDriver.h is and everything should work.

Lytt-2020, 2021-04-01:
I’m very sorry to tell you, I recompiled it according to what you said, but it’s still not possible. The installation path of my include/asynDriver.h is /home/lytt/epics/asyn4/asyn4-38/include/asynDriver.h, so ASYN=/home/lytt/epics/asyn4/asyn4-38 or to /asyn4- 38/include still reports an error. May I ask if there is something that is not installed before the error is reported. Thank you!

dirk-zimoch, 2021-04-01:
It should work like this.
Your StreamDevice installation is `/home/lytt/epics/streamdevice/StreamDevice-master`? (with for example `/home/lytt/epics/streamdevice/StreamDevice-master/src/StreamCore.cc`)
You configured ASYN in `/home/lytt/epics/streamdevice/StreamDevice-master/configure/RELEASE`?
Is there a directory `/home/lytt/epics/streamdevice/configure` ? That may be a problem.
What do you have in `/home/lytt/epics`? If you have `tree` installed, can you run `tree -d -L 3 -I 'O.*'` in `/home/lytt/epics` and mail the result to me?

Lytt-2020, 2021-04-01:
The installation directory of my StreamDevice is /home/lytt/epics/streamdevice/StreamDevice-master, and it is just like the example you mentioned. I have configured the installation path of ASYN in RELEASE, and the path of configure is /home/lytt/epics/streamdevice/StreamDevice-master/configure. The tree in my epics is as follows. Thank you!

├── asyn4
│   └── asyn4-38
│       ├── asyn
│       ├── bin
│       ├── configure
│       ├── db
│       ├── dbd
│       ├── documentation
│       ├── html
│       ├── include
│       ├── iocBoot
│       ├── lib
│       ├── makeSupport
│       ├── opi
│       ├── templates
│       ├── testApp
│       ├── testArrayRingBufferApp
│       ├── testAsynPortClientApp
│       ├── testAsynPortDriverApp
│       ├── testBroadcastApp
│       ├── testConnectApp
│       ├── testEpicsApp
│       ├── testErrorsApp
│       ├── testFtdiApp
│       ├── testGpibApp
│       ├── testGpibSerialApp
│       ├── testIPServerApp
│       ├── testManagerApp
│       ├── testOutputCallbackApp
│       ├── testOutputReadbackApp
│       └── testUsbtmcApp
├── base
│   ├── bin
│   │   └── linux-x86_64
│   ├── configure
│   │   ├── O.Common
│   │   ├── O.linux-x86_64
│   │   └── os
│   ├── db
│   ├── dbd
│   ├── documentation
│   ├── html
│   │   ├── freeList
│   │   ├── gpHash
│   │   └── image
│   ├── include
│   │   ├── compiler
│   │   └── os
│   ├── lib
│   │   ├── linux-x86_64
│   │   ├── perl
│   │   └── pkgconfig
│   ├── src
│   │   ├── ca
│   │   ├── ioc
│   │   ├── libCom
│   │   ├── std
│   │   ├── template
│   │   └── tools
│   ├── startup
│   ├── templates
│   │   ├── makeBaseApp
│   │   └── makeBaseExt
│   └── top
│       ├── bin
│       ├── configure
│       ├── db
│       ├── dbd
│       ├── firstIocApp
│       ├── include
│       ├── iocBoot
│       └── lib
├── calc
│   └── calc_R3-4
│       └── calc-3-4
├── extensions
│   ├── bin
│   │   └── linux-x86_64
│   ├── configure
│   │   ├── O.Common
│   │   ├── O.linux-x86_64
│   │   └── os
│   ├── lib
│   │   └── linux-x86_64
│   └── src
│       └── medm
├── ipac
│   └── ipac-2.15
│       ├── configure
│       ├── drvIpac
│       ├── drvTip810
│       ├── iocsh
│       ├── ip520
│       └── tyGSOctal
├── openssl
│   └── openssl-1.1.1j
│       ├── bin
│       ├── include
│       ├── lib
│       ├── share
│       └── ssl
├── openssl0
│   └── openssl-1.1.1j
│       ├── apps
│       ├── Configurations
│       ├── crypto
│       ├── demos
│       ├── doc
│       ├── engines
│       ├── external
│       ├── fuzz
│       ├── include
│       ├── lib
│       ├── ms
│       ├── os-dep
│       ├── ssl
│       ├── test
│       ├── tools
│       ├── util
│       └── VMS
├── rpcsvc
│   └── rpcsvc-proto-1.4.2
│       ├── m4
│       ├── po
│       ├── rpcgen
│       └── rpcsvc
├── streamdevice
│   └── StreamDevice-master
│       ├── config
│       ├── configure
│       ├── dbd
│       ├── docs
│       ├── include
│       ├── pcre
│       ├── src
│       └── streamApp
└── synApps
    └── synApps_6_1
        └── support

139 directories

dirk-zimoch, 2021-04-01:
I do not know why it does not work for you.
I can try to set up my own directory just as yours and see what happens. But not before next week.

Lytt-2020, 2021-04-01:
Okay, thank you for your attention! If you have any results, please email me in time. Thank you very much!

dirk-zimoch, 2021-04-27:
I have set up a system like yours and have absolutely no problem compiling everything.
I have `ASYN=/home/lytt/epics/asyn4/asyn4-38` in my `configure/RELEASE` as that is where you have asyn installed. (I also set `SUPPORT`, `CALC`, `PCRE` and `EPICS_BASE` but that's another story.)  I run `make` in `StreamDevice-master` and the include path shows up in the compiler arguments and StreamDevice has no problem finding `asynDriver.h`:
```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE      -D_X86_64_  -DUNIX  -Dlinux   -DUSE_TYPED_RSET
-DSTREAM_INTERNAL   -O3   -Wall    -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I../../include/compiler/gcc
-I../../include/os/Linux -I../../include      -I/home/lytt/epics/asyn4/asyn4-38/include
-I/usr/local/epics/base-7.0.5/include/compiler/gcc -I/usr/local/epics/base-7.0.5/include/os/Linux
-I/usr/local/epics/base-7.0.5/include        -MM -MF AsynDriverInterface.d  ../AsynDriverInterface.cc
```

 The only way I found to make asyn **not** show up in a `-I` option to the compiler as in your case is to spell the path wrong. If the path does not exist, then it will not show. Check again your `ASYN` setting.

Lytt-2020, 2021-04-28:
Thanks for your attention! This problem has been solved. As you said, it was caused by a path editing error. Thanks again for your reply! Thank you! And wish you a happy life!

closed 2021-04-28 by dirk-zimoch