paulscherrerinstitute/StreamDevice #75: STREAM_PROTOCOL_PATH separator

FreddieAkeroyd, 2021-06-22:
In https://github.com/paulscherrerinstitute/StreamDevice/commit/98fa595753bc9d071ecad828bc41d70ab04d1954 the default separator for `STREAM_PROTOCOL_PATH` was changed to be `;` on windows. This is more correct and avoids issues with referring to other drives like `d:` etc. but does mean that `STREAM_PROTOCOL_PATH` is no longer OS independent. I was wondering if a command to build a path in the right format could be added, so the `st.cmd` could then be
```
streamProtocolPath("dir1","dir2")
```
which would do an `epicsEnvSet` of  `STREAM_PROTOCOL_PATH` with the appropriate OS separator

dirk-zimoch, 2021-06-23:
A command to build paths would be a generally useful thing in EPICS. Actually, I have one at PSI: `addPath (ENVVAR, dir)`, but it is not for Windows (uses `:` only).

dirk-zimoch, 2021-06-23:
BTW: `addPath()` does more than simply concatinating the `dir`s. It pushes `dir` to the front, so that last added `dir` is found first, deletes any occurance that may already be in the variable, but always keeps an initial `.` in front, so that local files are always preferred to files in some other `dir`.
```
MYPATH=".:/a/b/c:/x/y:z"
addPath(MYPATH, "/new/dir")
MYPATH=".:/new/dir:/a/b/c:/x/y:z"
addPath(MYPATH, "/x/y")
MYPATH=".:/x/y:/new/dir:/a/b/c:z"
```

FreddieAkeroyd, 2021-06-23:
Your `addPath` would be a useful addition in EPICS base, but it would require stream device to use a newer EPICS base if you were to use it in examples. It is only an issue on windows though, I'm not sure how many people this affects. Would you prefer just an `addPath` in epics base over a `streamProtocolPath` like command, or maybe having both would be OK?

dirk-zimoch, 2021-06-23:
Is there a way to support both, `:` and `;` as path separators on Windows, without getting confused with drive letters?
Like "A single letter followed by `:` is a drive, not a directoy. (And `/` is `\`)"
So `cfg:c:/epics:\abc` would split to `cfg` `c:\epics` and `\abc`.
[I really do not like Windows.]

FreddieAkeroyd, 2021-06-23:
That would cover most cases, if somebody did create a directory called "c" that would just have to make sure it was referred to as` :./c:` or `:c/:` to distinguish it from `:c:\dir:`

dirk-zimoch, 2021-07-05:
Fixed in release 2.8.20.

closed 2021-07-05 by dirk-zimoch