paulscherrerinstitute/StreamDevice #79: Stream device 2.8.20 builds failure in Windows x86

FreddieAkeroyd, 2021-07-28:
Building Stream device 2.8.20 on `win32-x86` fails with the error
``` 
[2021-07-28T04:49:20.198Z] StreamEpics.cc
[2021-07-28T04:49:20.198Z] c:\instrument\apps\epics32\support\streamdevice\master\src\streamepics.cc(467) : error C2440: '=' : cannot convert from 'const char *(__stdcall *)(void)' to 'const char *(__cdecl *)(void)'
[2021-07-28T04:49:20.198Z]         This conversion requires a reinterpret_cast, a C-style cast or function-style cast
```
building on `windows-x64` works fine.

The relevant line is
```
StreamGetThreadNameFunction = epicsThreadGetNameSelf;
```

The issue is caused by `epicsThreadGetNameSelf` being defined as `epicsShareAPI` resulting in it using the `stdcall` rather than default  `cdecl` calling convention on x86, on x64 Microsoft removed `stdcall` and it was aliased to `cdecl` hence all compiles OK.

There are two options to fix this:
- modify  `StreamGetThreadNameFunction` signature to include `epicsShareAPI`, however any user defined alternative functions would also need to be declared `epicsShareAPI` too
- create and assign a  `streamGetThreadNameDefaultFunction()` wrapper function that just calls `epicsThreadGetNameSelf()` 

@dirk-zimoch I am happy to submit a PR, I just wanted to check which your preference was

dirk-zimoch, 2021-08-23:
I guess I will go with alternative 2.
I am really not happy with the permanent Windows dll specific problems. Maybe I should drop Windows support to make my life easier?

dirk-zimoch, 2021-09-01:
fixed in commit dfbd308.

FreddieAkeroyd, 2021-09-04:
Thanks for fixing - I am happy to help with resolving any windows specific issues, having stream device supported on windows is important for us. I'm assuming EPICS has `epicsShareAPI` set to `__stdcall` for compatibility with being called from programming languages other than C/C++, though a limitation of `__stdcall` is that it does not support variable argument functions hence some EPICS functions (like `errlogPrintf`) are not declared with `epicsShareAPI`.

closed 2021-09-28 by dirk-zimoch