paulscherrerinstitute/StreamDevice #83: RTEMS 5 patch for conflicting declaration of mystrncasecmp()

mrippa, 2022-03-30:
Hi Dirk,

Building with RTEMS5(gcc7.5), I found the following compile error.

```
/gem_base/targetOS/RTEMS/rtems/5/bin/powerpc-rtems5-g++ -B/gem_base/targetOS/RTEMS//rtems/5/powerpc-rtems5/mvme2307/lib/ -specs bsp_specs -qrtems     -mcpu=604 -mmultiple -mstring -mstrict-align -meabi       -D_GNU_SOURCE -D_DEFAULT_SOURCE            -DUNIX    -DUSE_TYPED_RSET -DSTREAM_INTERNAL   -O2 -g -ffunction-sections -fdata-sections   -Wall       -DMY_DO_BOOTP=NULL -DHAVE_PPCBUG -DNVRAM_INDIRECT    -D__LINUX_ERRNO_EXTENSIONS__ -DRTEMS_LEGACY_STACK   -I. -I../O.Common -I. -I. -I.. -I../../include/compiler/gcc -I../../include/os/RTEMS -I../../include   -I/gem_base/epics/support/sequencer/include   -I/gem_base/epics/support/asyn/include   -I/gem_base/epics/support/geminicalc/include   -I/gem_base/epics/support/sscan/include   -I/gem_base/epics/support/geminipcre/include -I/gem_base/epics/epics-base/include/compiler/gcc -I/gem_base/epics/epics-base/include/os/RTEMS -I/gem_base/epics/epics-base/include        -c ../ChecksumConverter.cc
../ChecksumConverter.cc:61:21: error: conflicting declaration of 'int mystrncasecmp(const char*, const char*, size_t)' with 'C' linkage
 #define strncasecmp mystrncasecmp
                     ^
../ChecksumConverter.cc:55:12: note: previous declaration with 'C++' linkage
 static int mystrncasecmp(const char *s1, const char *s2, size_t n)
            ^~~~~~~~~~~~~
```

The following patch looks like one way to fix it. This was not a problem on my Rocky 8 g++ 8.5. Details below.

```diff
diff --git a/src/ChecksumConverter.cc b/src/ChecksumConverter.cc
index 53ffcc6..2684d7d 100644
--- a/src/ChecksumConverter.cc
+++ b/src/ChecksumConverter.cc
@@ -52,11 +52,15 @@
 #if defined(vxWorks) || defined(__rtems__)
 // These systems have no strncasecmp at the moment
 // But avoid compiler errors in case strncasecmp exists in future versions
-static int mystrncasecmp(const char *s1, const char *s2, size_t n)
-{
-    int r=0;
-    while (n && (r = toupper(*s1)-toupper(*s2)) == 0) { n--; s1++; s2++; };
-    return r;
+extern "C" {
+
+    static int mystrncasecmp(const char *s1, const char *s2, size_t n)
+    {
+        int r=0;
+        while (n && (r = toupper(*s1)-toupper(*s2)) == 0) { n--; s1++; s2++; };
+        return r;
+    }
+
 }
 #define strncasecmp mystrncasecmp
 #endif
```

# Rocky 8 setup

## g++ rpm
```
$ rpm -qi gcc-c++-8.5.0-4.el8_5.x86_64
Name        : gcc-c++
Version     : 8.5.0
Release     : 4.el8_5
Architecture: x86_64
Install Date: Tue Mar 29 08:18:57 2022
Group       : Development/Languages
Size        : 32190812
License     : GPLv3+ and GPLv3+ with exceptions and GPLv2+ with exceptions and LGPLv2+ and BSD
Signature   : RSA/SHA256, Wed Nov 10 17:51:41 2021, Key ID 15af5dac6d745a60
Source RPM  : gcc-8.5.0-4.el8_5.src.rpm
Build Date  : Wed Nov 10 17:33:27 2021
Build Host  : ord1-prod-x86build003.svc.aws.rockylinux.org
Relocations : (not relocatable)
Packager    : infrastructure@rockylinux.org
Vendor      : Rocky
URL         : http://gcc.gnu.org
Summary     : C++ support for GCC
Description :
This package adds C++ support to the GNU Compiler Collection.
It includes support for most of the current C++ specification,
including templates and exception handling.
```

## g++
```
$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --disable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 8.5.0 20210514 (Red Hat 8.5.0-4) (GCC) 
```

# RTEMS 5

```
$ /gem_base/targetOS/RTEMS/rtems/5/bin/powerpc-rtems5-g++ -v
Using built-in specs.
COLLECT_GCC=/gem_base/targetOS/RTEMS/rtems/5/bin/powerpc-rtems5-g++
COLLECT_LTO_WRAPPER=/gem_base/targetOS/RTEMS/rtems/5/libexec/gcc/powerpc-rtems5/7.5.0/lto-wrapper
Target: powerpc-rtems5
Configured with: ../gcc-7.5.0/configure --prefix=/gem_base/targetOS/RTEMS/rtems/5 --bindir=/gem_base/targetOS/RTEMS/rtems/5/bin --exec_prefix=/gem_base/targetOS/RTEMS/rtems/5 --includedir=/gem_base/targetOS/RTEMS/rtems/5/include --libdir=/gem_base/targetOS/RTEMS/rtems/5/lib --libexecdir=/gem_base/targetOS/RTEMS/rtems/5/libexec --mandir=/gem_base/targetOS/RTEMS/rtems/5/share/man --infodir=/gem_base/targetOS/RTEMS/rtems/5/share/info --datadir=/gem_base/targetOS/RTEMS/rtems/5/share --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=powerpc-rtems5 --disable-libstdcxx-pch --with-gnu-as --with-gnu-ld --verbose --with-newlib --disable-nls --without-included-gettext --disable-win32-registry --enable-version-specific-runtime-libs --disable-lto --enable-newlib-io-c99-formats --enable-newlib-iconv --enable-newlib-iconv-encodings=big5,cp775,cp850,cp852,cp855,cp866,euc_jp,euc_kr,euc_tw,iso_8859_1,iso_8859_10,iso_8859_11,iso_8859_13,iso_8859_14,iso_8859_15,iso_8859_2,iso_8859_3,iso_8859_4,iso_8859_5,iso_8859_6,iso_8859_7,iso_8859_8,iso_8859_9,iso_ir_111,koi8_r,koi8_ru,koi8_u,koi8_uni,ucs_2,ucs_2_internal,ucs_2be,ucs_2le,ucs_4,ucs_4_internal,ucs_4be,ucs_4le,us_ascii,utf_16,utf_16be,utf_16le,utf_8,win_1250,win_1251,win_1252,win_1253,win_1254,win_1255,win_1256,win_1257,win_1258 --enable-threads --disable-plugin --enable-libgomp --enable-languages=c,c++
Thread model: rtems
gcc version 7.5.0 20191114 (RTEMS 5, RSB 5 (803d42cda7b3), Newlib 7947581) (GCC) 
```

mrippa, 2022-03-30:
Added PR #84

dirk-zimoch, 2022-03-30:
Thanks. I will have a look. Strange that it did not fail with my other architectures. I think I need to set up an RTEMS environment for better coverage of all possible EPICS systems.

dirk-zimoch, 2022-10-05:
Fixed.

closed 2022-10-05 by dirk-zimoch