paulscherrerinstitute/StreamDevice #91: capture into lsiRecord truncates

mdavidsaver, 2022-09-08:
I want to use a global `@mismatch` to capture some under documented error messages.

```
@mismatch {
    in "%(\$1LastErr-I)[^\r]";
}
```

I started with an lsiRecord

```
record(lsi, "$(P)LastErr-I") {
    field(DESC, "Last command error")
    field(SIZV, "128")
}
```

Unfortunately, the captured string is truncated at 39 characters.

```
record(waveform, "$(P)LastErr-I") {
    field(DESC, "Last command error")
    field(FTVL, "CHAR")
    field(NELM, "128")
}
```

However, an equivalent waveformRecord does not truncate.

This is with Base 7.0 and stream 94721c2b0e2ae118778d5783bd35cc642f573f60.

dirk-zimoch, 2022-09-09:
In redirections as you used it, StreamDevice acts according to the reported data type of the record. For lsi, that is STRING, which has 39 chars. You can see the same effect when trying to write to the lsi with dbpf <recordname> "<some extra long string>". That is exactly, what StreamDevice is doing when processing redirections.
To see the lsi as an array of chars, you need to access <recordname>.$
Again, you can check that behavior with dbpf and dbgf.
This should work as intended:
in "%(\$1LastErr-I.$)[^\r]";

dirk-zimoch, 2022-09-09:
BTW: The same is true when reading the lsi with CA.
That means, unlike other record types, using the root name of an lsi (abd probably lso as well) is pretty useless. You will always only be able to access the first 39 chars. You basically always have to use `.$`. This is quite unfortunate. 
Maybe that behavior should be documented in the lsi/lso documentation.

mdavidsaver, 2022-09-09:
Ah, yes.  You are correct.  And not the first person to remind me of the necessity of adding `.$` (aka. `.VAL$`) when using `lsi`.  There is no truncation if I append this in the proto file, and on my OPI screen (which I somehow didn't think of right away...).

> Maybe that behavior should be documented in the lsi/lso documentation ...

For what it is worth, I looked at your [lsi](https://paulscherrerinstitute.github.io/StreamDevice/lsi.html) page before opening this ticket.  So I would have seen a reminder there.

I am starting to wonder if `lsi` is worth my trouble vs `aai` or `waveform`.  It seems unlikely to me that I am the only one who continues to forget this subtlety.

closed 2022-09-09 by mdavidsaver