paulscherrerinstitute/StreamDevice #97: regex pseudo converter failing misbehaving?

hinxx, 2024-03-07:
I have a device sending an ASCII string with fields of different types (strings, chars, decimal integer, hex integer,..), delimited by ':'. I want to parse it out into several records. A sample input might look like this: `66:+44:255:0x122:1:R:190:A03`. FWIW, values '66', '+44' and '190' may change from one message to another; and take up 1 - 5 chars each.

If I do **not** use regex pseudo converter to convert 1 char value to a string all looks fine with this protocol:
```
in "%*i:%(\$1)d:%*i:0x%(\$2)03x:%(\$3)d:%(\$4)c:%(\$5)i:%(\$6)s";
```
`$4` is setting a value of a `stringin` record with a single char.

Then I discovered regex pseudo converter and hoped I could convert a single char into more representative string and set a value of `mbbi` record, instead.
Here is the protocol that does that:
```
in "%#/:R:/:Idle:/%#/:M:/:Ramping:/%#/:F:/:Fault:/%*i:%(\$1)d:%*i:0x%(\$2)03x:%(\$3)d:%(\$4)[a-zA-Z0-9]:%(\$5)i:%(\$6)s";
```
It works .. most of the time (no errors reported), but quite often I do see an error:
```
2024/03/07 10:39:16.783738 A03 Status: Input "A03" does not match format "%s"
```
`Status` PV is being processed every 2 s and sometimes the error happens and sometimes not. The error is always about the last token that needs to be parsed.

I've attached two dumps of debug messages, first one for successfully parsed string and the second one for the failed.
[stream-parsing.txt](https://github.com/paulscherrerinstitute/StreamDevice/files/14522373/stream-parsing.txt)

From what I can tell the buffer that regex touches seems to overflow (sometimes?). For example this would be reported in debug for failing case:
```
buffer after: A03:+11:255:0x122:1:F:12585:A033<00><00><00><00><00><00><00><00>
```
while this is seen in case of success:
```
buffer after: A03<00><00><00><00><00><00><00><00>
```

dirk-zimoch, 2024-03-07:
If you simply want to have more meaningful strings in an mbbi record, I suggest to use the %{} enum converter. It reads (the ugly) strings, and converts them to integers. Then the mbbi can take that integer and convert it to "nice" strings.
E.g. `%{R|M|F}` together with an mbbi with `ZRST="Idle"`, `ONST="Ramping"` and `TWST="Fault"` should give you the conversion you need.
When using regular expression substitution, keep in mind that it converts the whole rest of the input. That means, if you have `:R:` ***anywhere*** in your input, it will be replaced with `:Idle:`. Make sure that this has no unintended side effects.
The error confuses me, because "A03" *should* match `%s`. I will have a closer look at the debug output...

dirk-zimoch, 2024-03-07:
Maybe `StreamBuffer::replace` has a bug if the inserted string is longer than the removed one. I am checking this right now...

dirk-zimoch, 2024-03-07:
Can't find anything wrong. So this may take a while...

dirk-zimoch, 2024-03-08:
Thanks a lot! I will have a look.

dirk-zimoch, 2024-03-18:
I think I found the actual bug.https://github.com/paulscherrerinstitute/StreamDevice/blob/211f689cdf8acaf45a53fe035a1b1d0fac8a40d7/src/StreamBuffer.cc#L277
This needs to be a `<` instead of `<=`.
A `<=` may not leave the terminating `0x00` intact if the new content would just fit the buffer capacity, because the terminating `0x00` is not included in `len`/`newlen`.
Can you test if that fixes your problem?
(And the following `//` comment should be in the `else` block. I will correct that, too.)
I will push an update once you confirm that is fixes the problem.
Thank you for your debugging and analysis which lead me to this!

dirk-zimoch, 2024-03-19:
Found another, related bug in https://github.com/paulscherrerinstitute/StreamDevice/blob/211f689cdf8acaf45a53fe035a1b1d0fac8a40d7/src/StreamBuffer.cc#L289
This needs to use `len+offs` instead of `len` because content is moved from `offs` to `0`.

hinxx, 2024-03-19:
> This needs to use len+offs instead of len because content is moved from offs to 0.

You mean like this?

```
if (newlen<len) memset(buffer+newlen, 0, len+offs-newlen);
```

dirk-zimoch, 2024-03-19:
`if (newlen<len+offs) memset(buffer+newlen, 0, len+offs-newlen);`

hinxx, 2024-03-19:
Got it. Changed and running the test. Preliminary results look good, let me run this for a couple of hours.

hinxx, 2024-03-20:
I've been with running the fix over the night and I can confirm that the above two changes solve the issue for me.
Thank you!

dirk-zimoch, 2024-03-20:
Fixed in commit 1496089 tagged 2.8.25.

closed 2024-03-20 by dirk-zimoch