paulscherrerinstitute/StreamDevice #99: compilation problem with 2-8-25

LeeYangLBLBCS, 2024-04-22:
````
I upgraded StreamDevice from 2-8-23 to 2-8-25, the compilation failed:
/usr/bin/ar -rc    libstream.a     DebugInterface.o DummyInterface.o AsynDriverInterface.o EnumConverter.o BCDConverter.o RawConverter.o RawFloatConverter.o BinaryConverter.o ChecksumConverter.o MantissaExponentConverter.o TimestampConverter.o RegexpConverter.o devaoStream.o devaiStream.o devboStream.o devbiStream.o devmbboStream.o devmbbiStream.o devmbboDirectStream.o devmbbiDirectStream.o devlongoutStream.o devlonginStream.o devstringoutStream.o devstringinStream.o devwaveformStream.o devaaiStream.o devaaoStream.o devcalcoutStream.o devlsiStream.o devlsoStream.o devint64inStream.o devint64outStream.o devscalcoutStream.o StreamVersion.o StreamBuffer.o StreamError.o StreamProtocol.o StreamFormatConverter.o StreamCore.o StreamBusInterface.o StreamEpics.o
/usr/bin/ranlib libstream.a
make[2]: *** No rule to make target '../../lib/linux-x86_64/libpcre.a', needed by 'libstream.so'.  Stop.
make[2]: Leaving directory '/opt/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-25/src/O.linux-x86_64'
make[1]: *** [/opt/epics/base-7.0.4/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/opt/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-25/src'
make: *** [/opt/epics/base-7.0.4/configure/RULES_DIRS:85: src.install] Error 2
========================================
What's the required support packages for this version?
this is what I'm using:
ALIVE=$(SUPPORT)/alive-R1-1-1
ASYN=$(SUPPORT)/asyn-R4-44-2
AUTOSAVE=$(SUPPORT)/autosave-R5-10
BUSY=$(SUPPORT)/busy-R1-7-2
CALC=$(SUPPORT)/calc-R3-7-4
CAMAC=$(SUPPORT)/camac-R2-7-1
CAPUTRECORDER=$(SUPPORT)/caputRecorder-R1-7-3
DAC128V=$(SUPPORT)/dac128V-R2-9
DELAYGEN=$(SUPPORT)/delaygen-R1-2-1
DEVIOCSTATS=$(SUPPORT)/iocStats-3-1-16
IP=$(SUPPORT)/ip-R2-20-1
MODBUS=$(SUPPORT)/modbus-R3-3
SSCAN=$(SUPPORT)/sscan-R2-11-3
STD=$(SUPPORT)/std-R3-6-1
SNCSEQ=$(SUPPORT)/seq-2-2-9

dirk-zimoch, 2024-04-23:
The error message tells you what is missing:
`No rule to make target '../../lib/linux-x86_64/libpcre.a'`
It seems you are using a local PCRE library but have not yet built it.
You can use either a system PCRE installation if your system provides one. For example on Red Hat Linux you can use the pcre-devel rpm. In that case set `PCRE_LIB` and `PCRE_INCLUDE` in your RELEASE file to the locations of you PCRE installation (e.g. /usr/lib64).
Or you can use a locally built pcre (built as an EPICS support app). In that case set `PCRE` in your RELEASE file to the location of that support module. But that you have to build it first. See https://paulscherrerinstitute.github.io/StreamDevice/setup.html.

LeeYangLBLBCS, 2024-04-23:
sorry, I forgot that I've always compiled without pcre.

closed 2024-04-23 by LeeYangLBLBCS