paulscherrerinstitute/StreamDevice #100: help with compiling on windows 10 for the latest StreamDevice

LeeYangLBLBCS, 2024-05-15:
Does the latest StreamDevice compile on Windows 10?
I cloned the latest code and got the following errors when compiling:
=========================================================
gmake[2]: Entering directory 'C:/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-25/streamApp/O.windows-x64-static'
link -nologo   -incremental:no -opt:ref -release  -MACHINE:X64              -out:streamApp.exe    streamApp_registerRecordDeviceDriver.obj streamAppMain.obj     ../../lib/windows-x64-static/stream.lib  C:/epics/modules/synApps_6_1_epics7/support/asyn-R4-44-2/lib/windows-x64-static/asyn.lib  C:/epics/modules/synApps_6_1_epics7/support/calc-R3-7-4/lib/windows-x64-static/calc.lib  C:/epics/base-7.0.4/lib/windows-x64-static/dbRecStd.lib  C:/epics/base-7.0.4/lib/windows-x64-static/dbCore.lib  C:/epics/base-7.0.4/lib/windows-x64-static/ca.lib  C:/epics/base-7.0.4/lib/windows-x64-static/Com.lib netapi32.lib ws2_32.lib advapi32.lib user32.lib kernel32.lib winmm.lib dbghelp.lib
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkAddInput referenced in function init_record
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkAddOutput referenced in function init_record
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkClear referenced in function special
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkConnectionStatus referenced in function pvSearchCallback
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkGet referenced in function process
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkPut referenced in function execOutput
calc.lib(swaitRecord.obj) : error LNK2019: unresolved external symbol recDynLinkPutCallback referenced in function execOutput
streamApp.exe : fatal error LNK1120: 7 unresolved externals
gmake[2]: *** [C:/epics/base-7.0.4/configure/RULES_BUILD:213: streamApp.exe] Error 1120
gmake[2]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-25/streamApp/O.windows-x64-static'
gmake[1]: *** [C:/epics/base-7.0.4/configure/RULES_ARCHS:58: install.windows-x64-static] Error 2
gmake[1]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-25/streamApp'
gmake: *** [C:/epics/base-7.0.4/configure/RULES_DIRS:85: streamApp.install] Error 2

dirk-zimoch, 2024-05-15:
I am not familiar with the peculiarities if Windows linking. I usually use Linux and dynamic linking.
The missing symbols are from the SynApps calc module, I think. Maybe the library needs to be added to the linker command. (On Linux with dynamic linking, it is sufficient to add the calc module to the linker command for the stream library. I don‘t need to add it to the linker command for the app as well. But maybe Windows or/and static linking needs it.)

xiaoqiangwang, 2024-05-15:
These symbols are part of _sscan_ module. It is a indirect dependency via _calc_ module. Try to define it in your RELEASE file.

dirk-zimoch, 2024-05-15:
Looking into streamApp/Makefile, I find:
```
ifneq ($(words $(CALC) $(SYNAPPS)), 0)
# With synApps scalcout record
streamApp_DBD += calcSupport.dbd
PROD_LIBS += calc
ifneq ($(words $(SSCAN) $(SYNAPPS)), 0)
PROD_LIBS += sscan
endif
endif
```
Thus, as soon as you have `CALC` and `SSCAN` in your RELEASE file (and have built the two modules before building StreamDevice) it should work, just as Xiaoqiang suggested.

closed 2025-01-20 by dirk-zimoch