paulscherrerinstitute/StreamDevice #101: CRC adding

vandammeLPC, 2025-01-10:
Hello,

I am a beginner in this type of approach and I don't know if I am making my request in the right place.
Please excuse me if this is not the case.

I would like to know if it would be possible for you to add a CRC calculation method to the official Streamdevice distribution.
This calculation method is used in serial communication with [Spellmann MPS HV supplies](https://www.spellmanhv.com/-/media/en/Technical-Resources/Manuals/MPS-Communication-Protocol.pdf).

My checksum calculation function:

```
// Checksum used by Spellman High Voltage Supplies MPS
static uint32_t hv_mps(const uint8_t* data, size_t len, uint32_t sum)
{
    long int xsum;
    while (len--)
        {
        sum += *data++;
        }
    xsum = -(sum %256);
    sum = xsum & 0x7F;
    sum |= 0x40;
    return sum;
}
```
```
...
static checksum checksumMap[] =
..
    {"hv_mps",  hv_mps,           0x00,       0x00,       1}  // 0x63
..
```
Thank you in advance

dirk-zimoch, 2025-01-10:
Looks straight forward enough. Unless I find an unexpected problem with it, I will add it to the next release.  I may change some implementation details, like: `-(sum %256) & 0x7F | 0x40 = -(sum & 0xFF) & 0x7F | 0x40 = (-sum) & 0x7F | 0x40`
Actually, the documentation asks for `(-sum) & 0xFF & 0x7F | 0x40 =  (-sum) & 0x7F | 0x40`.

vandammeLPC, 2025-01-10:
Ok no worries for your implementation, as long as it works.
I have been using this method for several years now.
Thanks