paulscherrerinstitute/StreamDevice #104: asynDisconnected status

bfrk, 2025-06-13:
We have a problem with a device that sends (only) unsolicited data (on a serial interface, accessed via terminal server). We process the record via SCAN="I/O Intr" and we have this simple protocol file:
```
ReadTimeout = 1200;
PollPeriod = 100;
InTerminator = "#";
ExtraInput = Ignore;

inDisplay {
  in "DISPLAY:=%dBA:=%(\$1)d";
}
```
When the terminalserver becomes disconnected the IOC issues messages at a rate that, at least to me, looks like "as fast as possible". Here is an excerpt:
```
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618156 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618165 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618195 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618203 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618212 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618236 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618257 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618272 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618301 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618313 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618328 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618358 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618369 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618377 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618408 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618423 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618435 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618459 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618472 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618480 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618506 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618514 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618524 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jan 27 13:08:49 2025]2025/01/27 13:08:49.618549 RADDZ4R RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
```
Note how the timestamps from streamDevice are all within the same millisecond(!)

I have looked at the interface code between asyn and streamDevice. While I have no explanation for the (apparent) endless loop here, I did notice that the `asynDisconnected` status is returned only by asyn's `queueRequest` method, whereas streamDevice seems to expect it being returned from the `read` and `write` methods.

dirk-zimoch, 2025-06-13:
You have not seen a message from the exceptionHandler, have you? (With debug output turned on.)

dirk-zimoch, 2025-06-13:
You can try to add in `readHandler()`, `case asynError:` the line
`pasynManager->isConnected(pasynUser, &connected);`
just before `if (!connected)`. Maybe the same in `writeHandler()` too.
Then the `connected` status would not rely on the `exceptionHandler()` updating `conencted` before the `read()` or `write()` call returns.

bfrk, 2025-06-13:
At the moment I do not have a test setup for this, sorry for that, which means I cannot easily play with parameters ATM. I will try to create one next week.

base: R3-15-8
asyn: R4-45
streamDevice: R2-8-24

The terminal server is a MOXA, it is accessed via plain TCP/IP. This is from the startup script:

drvAsynIPPortConfigure RADDZ4R serialc1s15g.ctl.bessy.de:4004

> You have not seen a message from the exceptionHandler, have you? (With debug output turned on.)

The excerpt is from a failure a few months ago. I chose it specifically because it does not have any other messages in it, so it is easier to see what is going on. In the meantime more devices have been added (and perhaps asyn settings, too). The most recent failure has more messages but is pretty confusing, here is another excerpt:

```
[Mon Jun  2 14:21:32 2025]RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478815 RADDZ2B RADDZ2B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478831 RADDZ4R RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478860 RADDZ2B RADDZ2B:inVal: asynError in read: serialc1s15g.ctl
.bessy.de:4002 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478871 RADDZ4R RADDZ4R:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478877 RADDZ3B RADDZ3B:inVal: I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478951 RADDZ3B RADDZ3B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478934 2025/06/02 14:21:32.478933 RADDZ4R RADDZ2B RADDZ2B:inVal: 
I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479001 RADDZ2B RADDZ2B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478821 RADDZ1B RADDZ1B:inVal: I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.478984 RADDZ3B RADDZ3B:inVal: asynError in read: serialc1s15g.ctl
.bessy.de:4003 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479033 RADDZ1B RADDZ1B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479050 2025/06/02 14:21:32.479083 RADDZ2B RADDZ4R RADDZ4R:inVal: 
I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479090 RADDZ1B RADDZ1B:inVal: asynError in read: serialc1s15g.ctl
.bessy.de:4001 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479128 RADDZ4R RADDZ2B:inVal: asynError in read: serialc1s15g.ctl
.bessy.de:4002 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479171 RADDZ2B RADDZ2B:inVal: I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]RADDZ4R:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479150 2025/06/02 14:21:32.479108 RADDZ1B RADDZ1B:inVal: I/O erro
r after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]RADDZ3B 2025/06/02 14:21:32.479190 2025/06/02 14:21:32.479220 RADDZ4R RADDZ3B:inVal: 
I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]RADDZ2B RADDZ4R:inVal: asynError in read: serialc1s15g.ctl.bessy.de:4004 disconnected
:
[Mon Jun  2 14:21:32 2025]RADDZ2B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479263 RADDZ3B RADDZ3B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479303 RADDZ2B RADDZ2B:inVal: asynError in read: serialc1s15g.ctl
.bessy.de:4002 disconnected:
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479323 RADDZ2B RADDZ2B:inVal: I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479279 2025/06/02 14:21:32.479234 2025/06/02 14:21:32.479344 RADD
Z2B RADDZ2B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]RADDZ4R RADDZ1B 2025/06/02 14:21:32.479396 RADDZ3B RADDZ3B:inVal: asynError in read: 
serialc1s15g.ctl.bessy.de:4003 disconnected:
[Mon Jun  2 14:21:32 2025]RADDZ4R:inVal: I/O error after reading 0 bytes: ""
[Mon Jun  2 14:21:32 2025]RADDZ1B:inVal: Protocol aborted
[Mon Jun  2 14:21:32 2025]2025/06/02 14:21:32.479393 RADDZ2B RADDZ2B:inVal: asynError in read: serialc1s15g.ctl
.bessy.de:4002 disconnected:
```

This particular kind of device is the only one where we have observed this behavior.

I will try your suggestion next week. Thanks!

bfrk, 2025-06-17:
I agree with your analysis. This is really quite mysterious.

Some more data: terminating the TCP connection with killcx is not needed to reproduce. I can reliably get the observed behavior by unplugging the MOXA (network cable) and then hitting Ctrl-C. My guess is that what we have been observing in production was triggered when, after about 2 hours of inactivity, the TCP connection finally gets shut down.

This is what the output looks like with `asynSetTraceMask DOSI1V -1 0x0031` (all relevant bits set):
```
2025/06/17 16:37:56.243 box12.acc.bessy.de:4001 read.
2025/06/17 16:37:56.243 DOSI1V read from low-level driver returned 3
2025/06/17 16:37:56.242897 DOSI1V DOSI1V:inVal: asynError in read: box12.acc.bessy.de:4001 disconnected:
2025/06/17 16:37:56.242907 DOSI1V DOSI1V:inVal: I/O error after reading 0 bytes: ""
2025/06/17 16:37:56.242917 DOSI1V DOSI1V:inVal: Protocol aborted
2025/06/17 16:37:56.243 DOSI1V set Eos 0

2025/06/17 16:37:56.243 DOSI1V addr -1 queueRequest priority 0 not lockHolder
2025/06/17 16:37:56.243 asynManager::portThread port=DOSI1V callback
2025/06/17 16:37:56.243 DOSI1V get Eos 0

2025/06/17 16:37:56.243 DOSI1V set Eos 1

2025/06/17 16:37:56.243 box12.acc.bessy.de:4001 read.
2025/06/17 16:37:56.243 DOSI1V read from low-level driver returned 3
2025/06/17 16:37:56.242975 DOSI1V DOSI1V:inVal: asynError in read: box12.acc.bessy.de:4001 disconnected:
2025/06/17 16:37:56.242985 DOSI1V DOSI1V:inVal: I/O error after reading 0 bytes: ""
2025/06/17 16:37:56.242997 DOSI1V DOSI1V:inVal: Protocol aborted
2025/06/17 16:37:56.243 DOSI1V set Eos 0

2025/06/17 16:37:56.243 DOSI1V addr -1 queueRequest priority 0 not lockHolder
2025/06/17 16:37:56.243 asynManager::portThread port=DOSI1V callback
2025/06/17 16:37:56.243 DOSI1V get Eos 0

2025/06/17 16:37:56.243 DOSI1V set Eos 1
```

BTW, note that one or two seconds after disconnecting the plug, I see the same messages from asyn (but NOT those from streamDevice), albeit in the expected rate i.e. one cycle every 100ms (the PollPeriod). It is also the case that after plugging the cable back in, everything recovers just fine (via asyn auto-reconnect).

dirk-zimoch, 2025-06-27:
Neither drvAsynIPPort.c nor `epicsSocketCreate()` sets socket option `SO_KEEPALIVE` and Linux does not enable it by default.Thus it is not surprising that we never see the socket closing. Thus, I have no idea how to reproduce what you are seeing.

dirk-zimoch, 2025-06-27:
Shouldn't calling `exit` from the ioc shell do the same? At least it does not trigger strange things. But I will try to register `epicsExit` and see what happens then.

bfrk, 2025-07-01:
I am baffled. I will try to create a self-contained example and see if I can still reproduce it here. If yes, will send it to you.

Thanks for your efforts and help!

dirk-zimoch, 2025-07-01:
Once difference is that I have tested with base 7.0.9 because I had no base 3.15 installed currently. I will try again with 3.15.8.

bfrk, 2025-07-07:
Thanks for the updates. Yes, the IOC terminates (after producing a few thousand error messages at a very high rate). Also, yes, by default `epicsExit` is not called on Linux (probably because the normal process `exit` already cleans up resources quite well).

dirk-zimoch, 2025-07-22:
Ok, keeping this open until further notice.