paulscherrerinstitute/StreamDevice #106: windows: isspace debug assertion

FreddieAkeroyd, 2025-08-05:
A debug build on windows adds some additional checks to the standard library `isspace` function in that the integer argument `c` must satisfy  `c >= -1 && c <= 255`    I am seeing a debug assertion at https://github.com/paulscherrerinstitute/StreamDevice/blob/f15837e1c518fdac70f9bc7129538b91b5a28477/src/StreamFormatConverter.cc#L526 which i believe is being caused by the input character being `0xc2` which as it is stored locally as a `signed char` being turned into the integer `0xffffffc2`.  Casting to `unsigned char` before calling `isspace` would solve the issue, potentially generally by creating and using a local stream device `isspace` function however i suspect that other character functions may have a similar check and also need adjusting.

dirk-zimoch, 2025-08-05:
`isspace()` is defined to have `int` arguments. The `*input` argument is a `char` (neither `signed` nor `unsigned`). On all "sane" architectures I have seen so far, passing a `char` argument to the functions from `ctype.h` worked just fine. On architectures where `char` is stored as a signed value (x86) as well as on architectures where `char` is stores as an unsigned value (ppc).

However, the Linux man page mentions indeed:  "These  functions  check  whether c, which must have the value of an unsigned char or EOF,..."
So the check is right. It must be cast to `unsigned char` first. I will do that.

If this throws an error here, then the impact may be far larger than just StreamDevice. I found several locations in EPICS base and in asyn where `char` variables are passed to `isspace()` without casting to `unsigned char` first. Some cast to `int` explicitly which is of course totally pointless. They all may trigger this error when given non-ASCII characters. Maybe that should be addressed too.

FreddieAkeroyd, 2025-08-05:
I expect there are a lot of potential places needing casts or some other mitigation - as as well as `isspace` likely `isalnum`, `islower`, `isascii`  etc all get used too in our various modules.